jnk0le/AVR-UART-lib #4: AVR-UART-lib && Atmega328

badeip, 2018-02-08:
Hi,

Sorry for the rather vague issue, but I am having some problems getting your library to work with AtmelStudio 7 and Atmega328.
Even a clean project with the example `multiple usart.c`, and the required headers fails with multiple undeclared warnings. Explicitly defining the prototypes just digs me deeper down a rabbit hole with subsequent errors.
Is this a known incompatibility?

Regards,

Petter.

jnk0le, 2018-02-08:
Could you paste whole compilation log?

badeip, 2018-02-08:
```
------ Build started: Project: usart, Configuration: Debug AVR ------
Build started.
Project "usart.cproj" (default targets):
Target "PreBuildEvent" skipped, due to false condition; ('$(PreBuildEvent)'!='') was evaluated as (''!='').
Target "CoreBuild" in file "C:\Program Files (x86)\Atmel\Studio\7.0\Vs\Compiler.targets" from project "c:\users\petter\Documents\Atmel Studio\7.0\usart\usart\usart.cproj" (target "Build" depends on it):
	Task "RunCompilerTask"
		Shell Utils Path C:\Program Files (x86)\Atmel\Studio\7.0\shellUtils
		C:\Program Files (x86)\Atmel\Studio\7.0\shellUtils\make.exe all --jobs 4 --output-sync 
		Building file: .././main.c
		Invoking: AVR/GNU C Compiler : 5.4.0
		.././main.c: In function 'main':
c:\users\petter\Documents\Atmel Studio\7.0\usart\usart\main.c(39,5): error: implicit declaration of function 'uart1_init' [-Werror=implicit-function-declaration]
		     uart1_init(BAUD_CALC(115200));
		     ^
c:\users\petter\Documents\Atmel Studio\7.0\usart\usart\main.c(48,5): error: implicit declaration of function 'uart1_puts_P' [-Werror=implicit-function-declaration]
		     uart1_puts_P("hello from flashed, usart 1\r\n"); // write string to usart buffer from flash memory // string is parsed by PSTR() macro
		     ^
c:\users\petter\Documents\Atmel Studio\7.0\usart\usart\main.c(49,5): error: implicit declaration of function 'uart1_puts_p' [-Werror=implicit-function-declaration]
		     uart1_puts_p(foo_string);
		     ^
c:\users\petter\Documents\Atmel Studio\7.0\usart\usart\main.c(55,5): error: implicit declaration of function 'uart1_gets' [-Werror=implicit-function-declaration]
		     uart1_gets(buffer, 25); // read at most 24 bytes from buffer (CR,LF will not be cut)
		     ^
c:\users\petter\Documents\Atmel Studio\7.0\usart\usart\main.c(80,9): error: implicit declaration of function 'uart1_putfloat' [-Werror=implicit-function-declaration]
		         uart1_putfloat(0.1337f);
		         ^
c:\users\petter\Documents\Atmel Studio\7.0\usart\usart\main.c(81,9): error: implicit declaration of function 'uart1_puts' [-Werror=implicit-function-declaration]
		         uart1_puts("\r\n");
		         ^
c:\users\petter\Documents\Atmel Studio\7.0\usart\usart\main.c(82,9): error: implicit declaration of function 'uart1_putstr' [-Werror=implicit-function-declaration]
		         uart1_putstr(buffer); // write array string to usart buffer
		         ^
		cc1.exe: some warnings being treated as errors
		make: *** [main.o] Error 1
		"C:\Program Files (x86)\Atmel\Studio\7.0\toolchain\avr8\avr8-gnu-toolchain\bin\avr-gcc.exe"  -x c -DDEBUG -DBOARD=ATMEGA328P_XPLAINED_MINI  -I"C:\Program Files (x86)\Atmel\Studio\7.0\Packs\Atmel\ATmega_DFP\1.2.203\include" -I"../src/ASF/common/boards" -I"../src/ASF/mega/utils/preprocessor" -I"../src/ASF/mega/utils" -I"../src/ASF/common/utils" -I"../src/ASF/mega/boards" -I"../src/ASF/mega/boards/atmega328p_xplained_mini" -I"../src/ASF/common/services/gpio" -I"../src/ASF/common/services/ioport" -I"../src" -I"../src/config" -I"../src/ASF/mega/drivers/cpu" -I"../src/ASF/common/services/clock" -I"../src/ASF/common/services/delay" -I"../src/ASF/mega/drivers/usart"  -O1 -fdata-sections -ffunction-sections -fdata-sections -fpack-struct -fshort-enums -g3 -Wall -mmcu=atmega328p -B "C:\Program Files (x86)\Atmel\Studio\7.0\Packs\Atmel\ATmega_DFP\1.2.203\gcc\dev\atmega328p" -c -std=gnu99 -fno-strict-aliasing -Wstrict-prototypes -Wmissing-prototypes -Werror-implicit-function-declaration -Wpointer-arith -mrelax -MD -MP -MF "main.d" -MT"main.d" -MT"main.o"   -o "main.o" ".././main.c" 
c:\users\petter\Documents\Atmel Studio\7.0\usart\usart\Debug\Makefile(133,1): error: recipe for target 'main.o' failed
	Done executing task "RunCompilerTask" -- FAILED.
Done building target "CoreBuild" in project "usart.cproj" -- FAILED.
Done building project "usart.cproj" -- FAILED.

Build FAILED.
========== Build: 0 succeeded or up-to-date, 1 failed, 0 skipped ==========
```

ziajek444, 2018-02-08:
Why van i read these messages?

jnk0le, 2018-02-08:
Code containing uart1_* calls should not compile at all for atmega 328P (not PB which has 2 usarts).
If you disable -Werror linker should complain about unresolved symbols.

On 328P both uart_* and uart0_* namings are valid.

badeip, 2018-02-08:
@jnk0le I hoped/thought that I could define multiple software UARTs for the atmega 328P, like arduino software serial that I am trying to phase out.
I am fairly new to atmega development, so it seems that I just exposed my ignorance ;)
I would really appreciate if you have any pointers to doing that, as my google fu has been insufficient the last couple of hours.
Thanks, and feel free to close this issue

jnk0le, 2018-02-09:
At the moment I'm only bug fixing and unweirdening this lib. 
Even if I meka a soft library it will be completely separated from this one.

closed 2018-02-09 by jnk0le