jnk0le/AVR-UART-lib #6: error: expected ‘;’, ‘,’ or ‘)’ before ‘char’

PaulMaxAvalosAguilar, 2018-06-28:
make[1]: Entering directory '/home/paul/Cprojects/Production/Encoder2/MicrocontrollerCode/AVR/Src_Modules/MicroC_Application'

avr-gcc -Wall -Os -DF_CPU=8000000 -mmcu=atmega328p -std=c11 -Wall  -D RELEASE  -c src/example.c  -o ../../Objects/ReleaseAvrStatic/example.o  
In file included from src/example.c:8:0:
src/usart.h:1593:27: warning: type defaults to ‘int’ in declaration of ‘__flash’
   void uart0_puts_p(const __flash char *string); // send string from flash memory
                           ^
src/usart.h:1593:35: error: expected ‘;’, ‘,’ or ‘)’ before ‘char’
   void uart0_puts_p(const __flash char *string); // send string from flash memory
                                   ^
src/example.c:12:6: warning: return type of ‘main’ is not ‘int’ [-Wmain]
 void main(void)
      ^
src/example.c: In function ‘main’:
src/example.c:24:2: warning: implicit declaration of function ‘uart0_puts_p’ [-Wimplicit-function-declaration]
  uart_puts_P("hello from flashed, usart\r\n"); // write string to usart buffer from flash memory // string is parsed by PSTR() macro
  ^
Makefile:25: recipe for target '../../Objects/ReleaseAvrStatic/example.o' failed
make[1]: *** [../../Objects/ReleaseAvrStatic/example.o] Error 1
make[1]: Leaving directory '/home/paul/Cprojects/Production/Encoder2/MicrocontrollerCode/AVR/Src_Modules/MicroC_Application'
Makefile:28: recipe for target 'Application' failed
make: *** [Application] Error 2

I think it is self explicative

jnk0le, 2018-06-29:
`void main()` is now fixed and the rest of the errors are related to `__flash` specifier that is supported by avr-gcc since 4.7 version.

jnk0le, 2018-07-10:
the problem with __flash specifier might also be the lack of suport in g++.
https://www.avrfreaks.net/forum/avr-g-lacks-flash

closed 2018-10-01 by jnk0le

hdelatorre21, 2019-01-07:
Hi, I am trying to use this library in Atmel Studio 7 but it seems that the attribute __flash is not compatible with it, I was using the test program that comes whit the library and this two errors below was the only thing that prevent the program to compile, I am quite new to programming in C, so maybe it's an easy problem to solve, any help or advice you can give me so I can use this library in Atmel Studio 7 would be very well received.

The two erros occur in each of the next functions
void uart1_puts_p(const __flash char *string); // send string from flash memory  
void uart2_puts_p(const __flash char *string); // send string from flash memory
void uart3_puts_p(const __flash char *string); // send string from flash memory

Errors:

Severity	Code	Description	Project	File	Line
Error		'__flash' does not name a type	UART_lib_TEST	C:\Users\De la Torre\Documents\Atmel Studio\C Programming\Programas AVR\UART_lib_TEST\UART_lib_TEST\usart.h	1675

Severity	Code	Description	Project	File	Line
Error		expected ',' or '...' before 'char'	UART_lib_TEST	C:\Users\De la Torre\Documents\Atmel Studio\C Programming\Programas AVR\UART_lib_TEST\UART_lib_TEST\usart.h	1675

jnk0le, 2019-01-07:
Looks like the fix only addressed uart0

hdelatorre21, 2019-01-08:
I think it's because the IDE (Atmel Studio 7) I'm using, maybe it's the same thing that used to happen with g++ until the 4.7 version as you metioned, I thought maybe there could be a easy fix to make it work in Atmel but if there's not, well I guess I'll just have to keep working.

Thankyou very much for your response.

commit 936b3559 (2018-10-01): fixed cpp compatibility

fixed #6